platformio/platformio-atom-ide #1446: C/C++ project index rebuild failed

a-tom-s, 2018-01-27:
### Verbose mode can be enabled via `-v, --verbose` option
System: ESP8266 80MHz RAM/80K Flash/4M
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe9' in position 131: ordinal not in range(128):
File "/home/thomas/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "/home/thomas/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/thomas/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 256:
call_stack[-1].globals)
File "/home/thomas/.platformio/platforms/espressif8266@src-428e6fe6a638295277b17b20ae60f081/builder/main.py", line 375:
target_elf = env.BuildProgram()
File "/home/thomas/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/thomas/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 63:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "/home/thomas/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/thomas/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 261:
env.ConvertInoToCpp()
File "/home/thomas/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/thomas/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "/home/thomas/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 54:
return self.process(contents)
File "/home/thomas/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
File "/home/thomas/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 88:
fp.write(contents)

flashy02, 2018-01-27:
I have this Problem too. Since the Update to platformio 2.0.0

ivankravets, 2018-01-27:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1438

closed 2018-01-27 by ivankravets