platformio/platformio-atom-ide #1448: C/C++ project index rebuild failed

chenhonglin, 2018-01-28:
Error: [01/28/18 16:28:09] Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
System: ESP8266 80MHz RAM/80K Flash/4M
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
Modes: Finder/chain Compatibility/light
Collected 49 compatible libraries
Scanning dependencies...
UnicodeDecodeError: 'ascii' codec can't decode byte 0xb5 in position 13: ordinal not in range(128):
File "C:\users\administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 256:
call_stack[-1].globals)
File "C:\users\administrator\.platformio\platforms\espressif8266\builder\main.py", line 353:
target_elf = env.BuildProgram()
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 69:
env.Append(LIBS=env.BuildProjectLibraries())
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 843:
project.search_deps_recursive()
File "C:\users\administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 411:
for inc in self._get_found_includes(search_files):
File "C:\users\administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 354:
self.env.File(path), self.env, tuple(include_dirs))
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Scanner\__init__.py", line 213:
node_list = self.function(node, env, path)
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Scanner\__init__.py", line 335:
return self.scan(node, path)
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Scanner\__init__.py", line 382:
n, i = self.find_include(include, source_dir, path)
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Scanner\__init__.py", line 409:
n = SCons.Node.FS.find_file(include[1], paths)
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 3487:
node, d = dir.srcdir_find_file(filename)
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 2036:
except KeyError: node = rdir.file_on_disk(filename)
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 2067:
if self.entry_exists_on_disk(name):
File "C:\Users\Administrator\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 1939:
result = os.path.exists(self._abspath + OS_SEP + name)
 [ERROR] Took 2.65 seconds

kioan, 2018-01-28:
Try to remove the folder C:\users\administrator.platformio and restart the IDE.

ivankravets, 2018-01-30:
Do you have non-ASCII chars in project path?

ivankravets, 2018-01-30:
Fixed in the latest PIO Core. Please open PIO IDE Terminal and type `pio upgrade --dev`

closed 2018-01-30 by ivankravets