platformio/platformio-atom-ide #1450: Create Your PIO Account - error

myk3y, 2018-01-29:
I just installed Atom and Platformio IDE Home 0.6.1·Core 3.5.2a5 on Mac OS X 10.13.2 

Everything seems to be working fine - can compile, create projects, import, etc., but when I try and git and create an account as prompted I get the following error: 

```
Invalid response: <html>
 <head><title>502 Bad Gateway</title></head>
 <body bgcolor="white">
 <center><h1>502 Bad Gateway</h1></center>
 <hr><center>nginx</center>
 </body>
 </html>
```

checking for update, restarting Atom, etc. doesn't solve the problem. And where the heck do you register for an account outside of the IDE?

myk3y, 2018-01-29:
Closed: need the "Use Proxy Settings When Calling APM" option enabled - I disabled it as I don't use network proxies.

closed 2018-01-29 by myk3y

ivankravets, 2018-01-29:
Sorry, it was our issue with a server. We have fixed that recently.