platformio/platformio-atom-ide #1451: C/C++ project index rebuild failed

gnkarn, 2018-01-29:
Error: [Mon Jan 29 12:13:46 2018] Processing sonoff (build_flags: -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512, lib_deps: PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON, platform: espressif8266, board: esp01_1m, framework: arduino, board_flash_mode: dout)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

================================== [SUMMARY] ==================================
Environment sonoff-ds18x20	[SKIP]
Environment sonoff-minimal	[SKIP]

TypeError: Tried to lookup Dir '/Users/Gus-nuevo/.platformio/platforms/espressif8266/builder' as a File.:
File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/builder/main.py", line 134:
env.SConscript("$BUILD_SCRIPT")
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/Users/Gus-nuevo/.platformio/platforms/espressif8266/builder/main.py", line 353:
target_elf = env.BuildProgram()
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 52:
env.ProcessFlags(env.get("BUILD_FLAGS"))
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 112:
parsed_flags = env.ParseFlags(str(flags))
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Environment.py", line 800:
do_parse(arg)
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Environment.py", line 730:
dict['LIBS'].append(self.fs.File(arg))
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Node/FS.py", line 1413:
return self._lookup(name, directory, File, create)
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Node/FS.py", line 1392:
return root._lookup_abs(p, fsclass, create)
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Node/FS.py", line 2413:
result.must_be_same(klass)
File "/Users/Gus-nuevo/.platformio/packages/tool-scons@2.4.1/script/../engine/SCons/Node/FS.py", line 652:
(self.__class__.__name__, self.get_internal_path(), klass.__name__))
========================== [ERROR] Took 0.80 seconds ==========================
Environment sonoff	[ERROR]
========================== [ERROR] Took 0.81 seconds ==========================

ivankravets, 2018-01-29:
Please provide a contents of `platformio.ini`.

Inspirati, 2018-02-17:
**I am experiencing a similar failure when trying to build the MavESP8266 project from the source at:** https://github.com/dogmaphobic/mavesp8266.git

**The full build output is:**
robert-Inspiron-5748:~/GitHub/dogmaphobic/mavesp8266 (master)$ platformio run
[Sat Feb 17 20:22:41 2018] Processing esp12e (build_flags: -Wl,-Tesp8266.flash.4m.ld, platform: espressif8266, upload_speed: 921600, board: esp12e, framework: arduino, extra_script: esp_extra.py)
------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
TypeError: Tried to lookup Dir '/home/robert/.platformio/platforms/espressif8266/builder' as a File.:
File "/home/robert/.atom/packages/platformio-ide/penv/local/lib/python2.7/site-packages/platformio/builder/main.py", line 142:
env.SConscript("$BUILD_SCRIPT")
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/home/robert/.platformio/platforms/espressif8266/builder/main.py", line 353:
target_elf = env.BuildProgram()
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/robert/.atom/packages/platformio-ide/penv/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 54:
env.ProcessFlags(env.get("BUILD_FLAGS"))
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/robert/.atom/packages/platformio-ide/penv/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 112:
parsed_flags = env.ParseFlags(str(flags))
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 800:
do_parse(arg)
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 730:
dict['LIBS'].append(self.fs.File(arg))
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Node/FS.py", line 1404:
return self._lookup(name, directory, File, create)
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Node/FS.py", line 1383:
return root._lookup_abs(p, fsclass, create)
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Node/FS.py", line 2403:
result.must_be_same(klass)
File "/home/robert/.platformio/packages/tool-scons/script/../engine/SCons/Node/FS.py", line 652:
(self.__class__.__name__, self.get_internal_path(), klass.__name__))
==================================================== [ERROR] Took 0.35 seconds ====================================================

ivankravets, 2018-02-17:
It works for me with `git clone --recursive https://github.com/dogmaphobic/mavesp8266.git`.

Please file any issues here https://github.com/dogmaphobic/mavesp8266/issues

closed 2018-02-17 by ivankravets