platformio/platformio-atom-ide #1452: C/C++ project index rebuild failed

Gianbacchio, 2018-01-29:
Error: [01/29/18 19:03:34] Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino; lib_extra_dirs: ~/Documents/03-Arduino/libraries)

Verbose mode can be enabled via `-v, --verbose` option
System: ESP8266 80MHz RAM/80K Flash/4M
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe0' in position 3984: ordinal not in range(128):
File "C:\Users\Davide\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Davide\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Davide\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 256:
call_stack[-1].globals)
File "C:\Users\Davide\.platformio\platforms\espressif8266\builder\main.py", line 357:
target_elf = env.BuildProgram()
File "C:\Users\Davide\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\Davide\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 63:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\Davide\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\Davide\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 261:
env.ConvertInoToCpp()
File "C:\Users\Davide\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\Davide\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "C:\Users\Davide\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 54:
return self.process(contents)
File "C:\Users\Davide\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
File "C:\Users\Davide\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 88:
fp.write(contents)
 [ERROR] Took 12.09 seconds

ivankravets, 2018-01-29:
Should be fixed recently. Could you open PIO IDE Terminal and type `pio upgrade --dev`. Does it work now?

Gianbacchio, 2018-01-29:
ok, try it now, thank you!