platformio/platformio-atom-ide #1453: Clicking "Open" does not open the project

kubark42, 2018-01-29:
Observed behavior: Nothing happens. All other PlatformIO home page buttons function normally.
Expected behavior: Project should open. There should be some level of feedback when project fails to open.

Environment:
Surface Book 2
Windows 10
Atom 1.23.3 x64
Python 2.7.14

```
PS C:\Users\Kenn Sebesta> pio upgradeYYou're up-to-date!
PlatformIO 3.5.2a5 is currently the newest version available.
PS C:\Users\Kenn Sebesta>
PS C:\Users\Kenn Sebesta> pio update
UUpdating tool-scons                      @ 3.30001.0
[Up-to-date]
Updating tool-unity                      @ 1.20302.1      [Up-to-date]
Updating contrib-pysite                  @ 0.1.4          [Up-to-date]
Updating contrib-piohome                 @ 0.6.1          [Up-to-date]
Updating tool-pioplus                    @ 0.13.4         [Up-to-date]

Platform Manager
================
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.6.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20400.0      [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

Library Manager
===============
PS C:\Users\Kenn Sebesta>
```

ivankravets, 2018-01-30:
Please open PIO IDE Terminal and type `pio home --port 8888`, then in browser please click on "Open". Do you see any errors?

kubark42, 2018-01-30:
> Please open PIO IDE Terminal and type pio home --port 8888, then in browser please click on "Open". Do you see any errors?

I have tried this before, but without defining the port. In any case, here is the pop-up window with both `pio home` and `pio home --port 8888`:
```
Project has been successfully configured and is located by this path: C:\Users\kenz\180104-103929-huzzah.
You can open it with your favourite IDE or process with platformio run command.
```

ivankravets, 2018-01-30:
Could you reproduce this issue even after restart?

kubark42, 2018-01-30:
@ivankravets I am unsure if you are referring to restart of computer, Atom, browser, or `pio`, but in all cases the issue always arises. It's as if the plugin simply isn't registering the click.

ivankravets, 2018-01-30:
Please open Atom: Menu> View > Developer > Toggle Developer Tools > Console.

Do you see any errors?

kubark42, 2018-01-30:
`null                          VM1084 18cd432….min.js:31`

And when I click there, it takes me to debugger:///VM1084 18cd4320383b16562325.min.js (gist: https://gist.github.com/kubark42/2699e4bc4bae129766c88e86e9766cb3). L31 is pretty epic (>65536 characters), so not really sure what of make of that.

ivankravets, 2018-06-21:
Please upgrade Atom and PIO package to the latest version.

closed 2018-06-21 by ivankravets

MarioJose, 2019-08-24:
I have the same problem.

Using GNU/Linux Debian Stretch 9
Atom 1.40.1 x64
PlatformIO 2.2.0 Core 4.0.2
Python 2.7.13 and 3.5.3

When click in open in project list, not happens. 

Console in Developer Tools not show error

When open terminal and run pio home, pio home is open in my default browser. When click to open project in list project I get this message (I get this message too when open project inside pio home in atom):

```
Project has been successfully configured and is located by this path: /home/user/Documents/PlatformIO/Projects/test.
You can open it with your favourite IDE or process with platformio run command.
```
and this erro in terminal:

```
Open PIO Home in your browser by this URL => http://127.0.0.1:8008
PIO Home has been started. Press Ctrl+C to shutdown.
Error: {'message': 'PIO Home IDE agent is not started', 'code': 4005}
```

Yes, I restarted Atom e linux.

ivankravets, 2019-08-24:
Atom is deprecated. Please use PlatformIO for VSCode.

MarioJose, 2019-08-24:
Is sad that PlatformIO is forgetting Atom and migrate totally to Microsoft VSCode.
But, that is...
Thank you, @ivankravets for you help

ivankravets, 2019-08-24:
It's not PlatformIO. Atom provides limited API for C/C++ development. Also, there are issues with PlatformIO IDE Terminal for Atom.

pfeerick, 2019-08-24:
@MarioJose This is not the same problem - just a similar symptom. This bug appears to be related to a minified javascript file. The issue we are seeing now is new, and affects both VSCode and Atom. Best to keep following platformio/platformio-vscode-ide/issues/975 and hopefully the fix applies to both, as it may actually be something from the pio core, not the extensions.

ivankravets, 2019-08-25:
See https://github.com/platformio/platformio-vscode-ide/issues/975#issuecomment-524649853