platformio/platformio-atom-ide #1461: C/C++ project index rebuild failed

navyatippa, 2018-01-31:
Error: [01/31/18 11:55:31] Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)

SCons import failed. Unable to find engine files in:
C:\Users\Naveen_atl\.platformio\packages\tool-scons\script\..\engine
C:\Users\Naveen_atl\.platformio\packages\tool-scons\script\scons-local-2.5.1
C:\Users\Naveen_atl\.platformio\packages\tool-scons\script\scons-local
c:\users\naveen_atl\.platformio\penv\scons-2.5.1
c:\users\naveen_atl\.platformio\penv\Lib\site-packages\scons-2.5.1
c:\users\naveen_atl\.platformio\penv\scons
c:\users\naveen_atl\.platformio\penv\Lib\site-packages\scons
Traceback (most recent call last):
File "C:\Users\Naveen_atl\.platformio\packages\tool-scons\script\scons", line 190, in <module>
import SCons.Script
File "C:\Users\Naveen_atl\.platformio\packages\tool-scons\script\..\engine\SCons\__init__.py", line 43, in <module>
import SCons.compat
ImportError: No module named compat
 [ERROR] Took 0.94 seconds

navyatippa, 2018-01-31:
what is the issue?
how to resolve help me

ivankravets, 2018-02-17:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1512

closed 2018-02-17 by ivankravets