platformio/platformio-atom-ide #1472: Uncaught Error: ENOENT: no such file or directory, lstat '~/Documents\PlatformIO\Projects\180202-...

SanUSB, 2018-02-02:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.3 x64
**Electron**: 1.6.15
**OS**: Unknown Windows version
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.1

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, lstat 'C:\Users\sandr\Documents\PlatformIO\Projects\180202-121016-esp12e\src\timerTIcker.ino.cpp'

```
At fs.js:902

Error: ENOENT: no such file or directory, lstat 'C:\Users\sandr\Documents\PlatformIO\Projects\180202-121016-esp12e\src\timerTIcker.ino.cpp'
    at fs.lstatSync (fs.js:902:18)
    at Object.fs.lstatSync (ELECTRON_ASAR.js:244:16)
    at Proxy.realpathSync (fs.js:1542:21)
    at Proxy.fs.realpathSync (ELECTRON_ASAR.js:321:29)
    at /packages/platformio-ide/lib/project/helpers.js:32:25
    at getActivePioProject (/packages/platformio-ide/lib/project/helpers.js:36:6)
    at /packages/platformio-ide/lib/main.js:170:19)
    at CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.23.3/resources/app/src/command-registry.js:388:36)
    at CommandRegistry.dispatch (~/AppData/Local/atom/app-1.23.3/resources/app/src/command-registry.js:273:23)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (~/AppData/Local/atom/app-1.23.3/resources/app/src/atom-environment.js:1393:34)
    at EventEmitter.outerCallback (~/AppData/Local/atom/app-1.23.3/resources/app/src/application-delegate.js:335:31)
    at emitThree (events.js:116:13)
    at EventEmitter.emit (events.js:194:7)
```

### Commands

```
     -2:10.7.0 intentions:highlight (input.hidden-input)
     -2:10.4.0 core:paste (input.hidden-input)
     -2:07.3.0 intentions:highlight (input.hidden-input)
     -2:07.0 core:undo (input.hidden-input)
     -1:56.9.0 editor:newline (input.hidden-input)
  6x -1:55.6.0 intentions:highlight (input.hidden-input)
     -1:55.1.0 core:paste (input.hidden-input)
 30x -1:09.7.0 core:move-down (input.hidden-input)
 37x -1:07.9.0 core:move-up (input.hidden-input)
 14x -0:59 core:move-down (input.hidden-input)
  6x -0:52.3.0 core:move-left (input.hidden-input)
     -0:43.4.0 intentions:highlight (input.hidden-input)
     -0:43.3.0 core:copy (input.hidden-input)
undefined -0:37.7.0 intentions:highlight (input.hidden-input)
     -0:25.4.0 core:paste (input.hidden-input)
     -0:07.3.0 platformio-ide:target:build (input.hidden-input)
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.16 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.1 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
tool-bar 1.1.4 
```

ivankravets, 2018-02-17:
Please project from project explorer and restart IDE

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets