platformio/platformio-atom-ide #1478: Uncaught Error: Could not start PIO Home server:

anishkgt, 2018-02-03:
HI Getting this error for a while since adding the platformio-ide

> [Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.3 x64
**Electron**: 1.6.15
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.1

### Stack Trace

Uncaught Error: Could not start PIO Home server: Error: Could not install 'contrib-pysite' with version requirements '>=0.1.2,<2' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

```
At C:\Users\georget\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1616

Error: Could not start PIO Home server: Error: Could not install 'contrib-pysite' with version requirements '>=0.1.2,<2' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1616:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:538:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at ChildProcess.cp.emit (/packages/platformio-ide/node_modules/cross-spawn/lib/enoent.js:34:29)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.16 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.1 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
tool-bar 1.1.4 
```

[image]
[image]

anishkgt, 2018-02-03:
tried several times delete the package folder and re-installing the package bit did not help. after installation I was asked to restart atom and then this keeps coming.

anishkgt, 2018-02-04:
hmmm so much for support !

ivankravets, 2018-02-05:
Uncaught Error: Could not start PIO Home server: Error: Could not install 'contrib-pysite' with version requirements '>=0.1.2,<2' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

ivankravets, 2018-02-17:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1171

closed 2018-02-17 by ivankravets