platformio/platformio-atom-ide #1480: C/C++ project index rebuild failed

BrianAker, 2018-02-04:
Error: Traceback (most recent call last):
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/platformio/commands/init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 38, in __init__
    self._gather_tplvars()
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 155, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 134, in __call__
    value = self.func(*args)
  File "/Users/brian/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 79, in get_project_build_data
    raise result.exception
AttributeError: 'NoneType' object has no attribute 'endswith'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2018-02-05:
Could you provide a content of `platformio.ini` file?

ivankravets, 2018-02-05:
Fixed in https://github.com/platformio/platformio-core/commit/1362630ed64d7812b69609ac176788a8a11de340. 

Please open PlatformIO IDE Terminal and type `pio upgrade --dev`. Try to build a project again.

closed 2018-02-05 by ivankravets