platformio/platformio-atom-ide #1482: C/C++ project index rebuild failed

pletchford, 2018-02-04:
Error: Traceback (most recent call last):
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\platformio\commands\init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 38, in __init__
    self._gather_tplvars()
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 155, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\platformio\util.py", line 134, in __call__
    value = self.func(*args)
  File "c:\users\mycomp~1\platfo~1\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 79, in get_project_build_data
    raise result.exception
TypeError: argument of type 'NoneType' is not iterable

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2018-02-17:
Please provide `platformio.ini` file

pletchford, 2018-02-17:
> <platformio-atom-ide@noreply.github.com>
> <author@noreply.github.com>
> rebuild failed (#1482)

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = 
https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
lib_extra_dirs = C:\Users\My Computer\.platformio\ArduinoLib
upload_port=192.168.1.68
lib_deps=
   https://github.com/adafruit/DHT-sensor-library

ivankravets, 2018-06-21:
Should be fixed in the latest release. Please update `platformio-ide` package via `Settings > Updates`.

closed 2018-06-21 by ivankravets