platformio/platformio-atom-ide #1484: C/C++ project index rebuild failed

pzaepfel, 2018-02-04:
Error: [02/04/18 21:14:07] Processing heltec_wifi_lora_32 (platform: espressif32; board: heltec_wifi_lora_32; framework: arduino; lib_extra_dirs: ~/Documents/Arduino/libraries)

Verbose mode can be enabled via `-v, --verbose` option
System: ESP32 240MHz RAM/288K Flash/1.25M
Converting precipitationSensorESP32_rev0.20.2_23012018.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
Modes: Finder/chain Compatibility/light
PlatformioException: Could not load broken JSON: C:\Users\pzaep\Documents\Arduino\libraries\GuiLibrary-master\library.json:
File "C:\Users\pzaep\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\pzaep\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\pzaep\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\pzaep\.platformio\platforms\espressif32\builder\main.py", line 163:
target_elf = env.BuildProgram()
File "C:\Users\pzaep\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\pzaep\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 69:
env.Append(LIBS=env.BuildProjectLibraries())
File "C:\Users\pzaep\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\pzaep\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 839:
lib_builders = env.GetLibBuilders()
File "C:\Users\pzaep\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\pzaep\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 771:
env, join(libs_dir, item), verbose=verbose)
File "C:\Users\pzaep\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 53:
env, path, verbose=verbose)
File "C:\Users\pzaep\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 104:
self._manifest = manifest if manifest else self.load_manifest()
File "C:\Users\pzaep\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 545:
manifest = util.load_json(join(self.path, "library.json"))
File "c:\users\pzaep\.platformio\penv\lib\site-packages\platformio\util.py", line 193:
"Could not load broken JSON: %s" % file_path)
 [ERROR] Took 1.61 seconds

ivankravets, 2018-02-17:
Please remove broken `C:\Users\pzaep\Documents\Arduino\libraries\GuiLibrary-master\library.json`

closed 2018-02-17 by ivankravets

pzaepfel, 2018-02-17:
Ok thanks2