platformio/platformio-atom-ide #1492: Problems with OTA on ESP8266 ESP-12E

paquechu, 2018-02-06:
Hello,
I have been with this problem for several months and I do not know how to solve it.
I have several ESP8266 ESP-12E modules to which I can not upload the firmware via OTA. Others instead I can.
I found this sketch (CheckFlashConfig.ino) to verify the characteristics of these modules. The result of the execution in the modules in which OTA fails is this:

```
Real flash id: 001570EF
Real size flash: 2097152

Flash ide size: 4194304
Flash ide speed: 40000000
Flash ide mode: DIO
Flash Chip configuration wrong!
```

In the modules that I can update via OTA "Flash real size" is equal to "Flash ide size" = 4194304 but in these "Flash real size" = 2097152, I imagine that this is the problem.

**Is there any possibility to update these modules by changing some parameter in the process of compiling or uploading the firmware?**

Thank you.

GanjaFarmer, 2018-02-08:
Hi, this may help: http://docs.platformio.org/en/latest/platforms/espressif8266.html#custom-flash-size

paquechu, 2018-02-08:
Hello,
I have read the entire document and also the Flash Layout link. Thank you very much :-)
I have done several tests changing the parameter build_flags with different values ​​here I put two cases:

`build_flags = -Wl, -Teagle.flash.2m.ld;`

and also

`build_flags = -Wl, -Teagle.flash.1m0.ld`

With these environment values:

```
[env: nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

```
The tests have been like this:

a) Test 1

- I Compiled the sketch without parameter build_flags
- If I try to update the firmware via OTA it fails with the error message

   [ERROR]: No response from the ESP

b) Test 2

- I Compiled the sketch with parameter build_flags = -Wl, -Teagle.flash.2m.ld;
- If I try to update the firmware via OTA it fails with the error message

   [ERROR]: No response from the ESP

c) Test 3

- I compiled the sketch with parameter build_flags = -Wl, -Teagle.flash.1m0.ld
- If I try to update the firmware via OTA, it works OK

In this last test no space is reserved for SPIFFS

In b) and c) approximately the same amount of memory is reserved. I could use the configuration of c) but then I lose 1MB of SPIFS ...

I do not understand why it works in c) and it does not work in b).

Test a) **KO**
```
eagle.flash.2m.ld
/ * Flash Split for 2M chips * /
/ * sketch 1019KB * /
/ * spiffs 1004KB * /
/ * eeprom 20KB * /

Real size flash: 2097152
sketchSize: 352816
sketchSpace: 692224
```

Test b) **OK**
```
eagle.flash.1m0.ld
/ * Flash Split for 1M chips, not SPIFFS * /
/ * sketch 999KB * /
/ * eeprom 20KB * /

Real size flash: 2097152
sketchSize: 352816
sketchSpace: 671744
```

Any idea why this may be happening?

Regards :-)

ivankravets, 2018-02-17:
http://community.platformio.org/

closed 2018-02-17 by ivankravets

ivankravets, 2019-01-09:
@buzzy please report issue here https://github.com/platformio/platform-espressif8266/issues and provide your `platformio.ini`.