platformio/platformio-atom-ide #1498: InstallationManager

gj23, 2018-02-08:
Error: Can not find Python Interpreter
    at PlatformIOCoreStage._callee8$ (C:\Users\whs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2478:23)
    at tryCatch (C:\Users\whs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:832:40)
    at Generator.invoke [as _invoke] (C:\Users\whs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1066:22)
    at Generator.prototype.(anonymous function) [as next] (C:\Users\whs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:884:21)
    at step (C:\Users\whs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1839:191)
    at C:\Users\whs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1839:361
    at process._tickCallback (internal/process/next_tick.js:103:7)

gj23, 2018-02-08:
i have try to install platformio-ide, but he cant find python 2.7 to complete. i have installed python 2.7 and set the variable in path on windows 10. python works correct...when i click on i have python and select the folder platformio installer won't accept it and is python still missing

ivankravets, 2018-02-17:
Duplicate of #1364

closed 2018-02-17 by ivankravets