platformio/platformio-atom-ide #1507: Uncaught Error: Could not start PIO Home server:

jmdearras, 2018-02-11:
new install

Uncaught Error: Could not start PIO Home server: 
Traceback (most recent call last): 
File "main.py", line 12, in init main 
File "c:\users\jimdea~2\platfo~1\penv\lib\site-packages\click__init__.py", line 18, in from .core import Context, BaseCommand, Command, MultiCommand, Group, \ 
File "c:\users\jimdea~2\platfo~1\penv\lib\site-packages\click\core.py", line 8, in from .types import convert_type, IntRange, BOOL File "c:\users\jimdea~2\platfo~1\penv\lib\site-packages\click\types.py", line 5, in from ._compat import open_stream, text_type, filename_to_ui, \ 
File "c:\users\jimdea~2\platfo~1\penv\lib\site-packages\click_compat.py", line 6, in from weakref import WeakKeyDictionary 
File "C:\Users\Jim De Aras.platformio\python27\Lib\weakref.py", line 14, in from _weakref import ( ImportError: cannot import name _remove_dead_weakref

C:\Users\Jim De Aras\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1616
Hide Stack Trace
Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "c:\users\jimdea~2\platfo~1\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\users\jimdea~2\platfo~1\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\users\jimdea~2\platfo~1\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\users\jimdea~2\platfo~1\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Users\Jim De Aras\.platformio\python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
    at C:\Users\Jim De Aras\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1616:23
    at ChildProcess.onExit (C:\Users\Jim De Aras\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:538:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at ChildProcess.cp.emit (C:\Users\Jim De Aras\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)

The error was thrown from the platformio-ide package. This issue has already been reported.

I ran the command in that thread.

Jim

jmdearras, 2018-02-11:
I tried:
"Navigate to System Control Panel and uninstall ALL Pythons from a system
Remove C:\Users\{user name}\.platformio folder
Restart IDE"
there were no other pythons, it re-installed platormIO but no go.

ivankravets, 2018-02-17:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1298

closed 2018-02-17 by ivankravets