platformio/platformio-atom-ide #1512: C/C++ project index rebuild failed

rmswimkktt, 2018-02-12:
Error: [02/12/18 13:46:35] Processing uno (platform: atmelavr; board: uno; framework: arduino)

SCons import failed. Unable to find engine files in:
C:\Users\maru\.platformio\packages\tool-scons\script\..\engine
C:\Users\maru\.platformio\packages\tool-scons\script\scons-local-2.5.1
C:\Users\maru\.platformio\packages\tool-scons\script\scons-local
c:\users\maru\.platformio\penv\scons-2.5.1
c:\users\maru\.platformio\penv\Lib\site-packages\scons-2.5.1
c:\users\maru\.platformio\penv\scons
c:\users\maru\.platformio\penv\Lib\site-packages\scons
Traceback (most recent call last):
File "C:\Users\maru\.platformio\packages\tool-scons\script\scons", line 190, in <module>
import SCons.Script
File "C:\Users\maru\.platformio\packages\tool-scons\script\..\engine\SCons\__init__.py", line 43, in <module>
import SCons.compat
ImportError: No module named compat
 [ERROR] Took 0.46 seconds

rmswimkktt, 2018-02-12:
I resolved.
Delete .platformio folder and reinstall PlatformIO.

closed 2018-02-12 by rmswimkktt

madleech, 2018-04-09:
I found that the permissions of `~/.platformio/packages/tool-scons/engine/SCons/` were messed up:

```
$ ls -l ~/.platformio/packages/tool-scons/engine/SCons
ls: SCons: Permission denied

$ sudo ls -l ~/.platformio/packages/tool-scons/engine/SCons
total 168
-rw-r--r--  1 michael  staff  84105  9 Apr 21:51 Environment.pyc

$ sudo chmod u=rwx ~/.platformio/packages/tool-scons/engine/SCons/

$ ls -l /Users/michael/.platformio/packages/tool-scons/engine/SCons
total 168
-rw-r--r--  1 michael  staff  84105  9 Apr 21:51 Environment.pyc
```

After `chmod`ing `~/.platformio/packages/tool-scons/engine/SCons/` with some more suitable permissions I get:
```
CorePackageManager: Installing tool-scons @ ~2.20501.4
Downloading...
Unpacking...
tool-scons @ 2.20501.4 has been successfully installed!
```

Success at last :)