platformio/platformio-atom-ide #1520: InstallationManager

LNRawz, 2018-02-15:
PIP: Exception:
Traceback (most recent call last):
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/basecommand.py", line 209, in main
    status = self.run(options, args)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/commands/install.py", line 310, in run
    wb.build(autobuilding=True)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/wheel.py", line 747, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 359, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 576, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/download.py", line 809, in unpack_url
    hashes=hashes
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/download.py", line 648, in unpack_http_url
    hashes)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/download.py", line 841, in _download_http_url
    stream=True,
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 480, in get
    return self.request('GET', url, **kwargs)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/download.py", line 377, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 468, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 576, in send
    r = adapter.send(request, **kwargs)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 46, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/sepje/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 447, in send
    raise SSLError(e, request=request)
SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)

LNRawz, 2018-02-15:
Problem here is that I am behind a firewall that does "SSL sniffing", and Atom sees the local AC root cert. I looked everywhere and could not find a place to make PlatformIO use the Atom conf (strict-ssl = false) or the even the Atom cacert.pem (where I did append the local root AC cert).
What else? Thankx.

jankap, 2018-06-19:
Same problem for me. Is it possible to provide custom CA certs?

ivankravets, 2018-06-21:
See https://stackoverflow.com/questions/25981703/pip-install-fails-with-connection-error-ssl-certificate-verify-failed-certi

closed 2018-06-21 by ivankravets