platformio/platformio-atom-ide #1523: InstallationManager

Wernerhi, 2018-02-15:
PIP: Error: spawn /Users/olaf 1/.platformio/penv/bin/pip ENOENT

Wernerhi, 2018-02-15:
<embedded>:121025 IntersectionObserver.observe(target): target element is not a descendant of root.
TreeView @ <embedded>:121025
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand platformio Array[1] Object
manager.js [sm]:82 Error: spawn platformio ENOENT
(anonymous) @ manager.js [sm]:82
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand conda Array[1] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand /Users/olaf 1/.platformio/penv/bin/python Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand python2.7 Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand virtualenv Array[3] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2493 User's Virtualenv: Error: spawn virtualenv ENOENT
_callee8$ @ /Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2493
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand python2.7 Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand /Users/olaf 1/.platformio/penv/bin/pip Array[4] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2226 PIP: Error: spawn /Users/olaf 1/.platformio/penv/bin/pip ENOENT
_callee5$ @ /Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2226
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand /Users/olaf 1/.platformio/penv/bin/pip Array[3] Object
utils.js [sm]:33 InstallationManager PIP: Error: spawn /Users/olaf 1/.platformio/penv/bin/pip ENOENT
notifyError @ utils.js [sm]:33
internal/process/next_tick.js:103 Uncaught (in promise) undefined
<embedded>:75145 Uncaught (in promise) Error: The workspace can only contain one instance of item [object Object]
    at ItemRegistry.module.exports.ItemRegistry.addItem (/Applications/Atom.app/Contents/Resources/app/src/item-registry.js:11:21)
    at PaneContainer.didAddPaneItem (/Applications/Atom.app/Contents/Resources/app/src/pane-container.js:269:29)
    at Pane.addItem (/Applications/Atom.app/Contents/Resources/app/src/pane.js:632:48)
    at Pane.activateItem (/Applications/Atom.app/Contents/Resources/app/src/pane.js:574:18)
    at Workspace.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/workspace.js:1175:20)
    at Generator.next (<anonymous>)
    at step (/Applications/Atom.app/Contents/Resources/app/src/workspace.js:11:279)

ivankravets, 2018-02-17:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1243

closed 2018-02-17 by ivankravets