platformio/platformio-atom-ide #1551: InstallationManager

VolodymyrGalushka, 2018-02-25:
PIP: Exception:
Traceback (most recent call last):
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 488, in get
    return self.request('GET', url, **kwargs)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/Users/vasudevdas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)

ivankravets, 2018-03-02:
@VolodymyrGalushka, @tigertim08, @Andorin, @mac-dj73, @dy008

Please open system terminal and provide an output of 
```
python -c "import ssl; print(ssl.OPENSSL_VERSION)"
```

ivankravets, 2018-03-02:
Try in system terminal
```
/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"
brew update
brew install openssl
```

After that, please remove `~/.platformio/penv` folder and restart IDE.

ivankravets, 2018-03-03:
Please upgrade PIO IDE to 2.1.0

closed 2018-03-03 by ivankravets

gitanupam, 2018-04-25:
Simply upgrading pip may also help but need to upgrade `pip` without using `pip` (because of recent TLS deprecation by Python.org sites). See this SO post for details: https://stackoverflow.com/q/49768770/1526703

mkaykisiz, 2018-06-03:
curl https://bootstrap.pypa.io/get-pip.py | python

commit 0d8e3bf7 (2018-03-03): Fix GitHub's "TLSV1_ALERT_PROTOCOL_VERSION" issue with PIO Core installer // Resolve #1551
commit 1eaf368e (2018-03-03): Merge branch 'release/v2.1.0'

* release/v2.1.0:
  Prepare 2.1.0 release
  Fix GitHub's "TLSV1_ALERT_PROTOCOL_VERSION" issue with PIO Core installer // Resolve #1551
  Multi-themes (Dark & Light) for PIO Home
  Fix clang code completion instructions link (#1536)