platformio/platformio-atom-ide #1570: InstallationManager

dy008, 2018-03-02:
PIP: Exception:
Traceback (most recent call last):
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 488, in get
    return self.request('GET', url, **kwargs)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/Users/dy/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:661)

dy008, 2018-03-02:
Macbook.Ver 10.13.3,Atom Ver 1.24.0 x64

ivankravets, 2018-03-02:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1551

closed 2018-03-02 by ivankravets

ivankravets, 2018-03-03:
Please upgrade PIO IDE to 2.1.0