platformio/platformio-atom-ide #1573: Uncaught TypeError: Cannot read property 'theme' of undefined

trupntrup, 2018-03-03:
[Enter steps to reproduce:]

1. launch Atom
2. launch PlatformIO Home

**Atom**: 1.24.0 x64
**Electron**: 1.6.16
**OS**: Mac OS X 10.13.3
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.1.0

### Stack Trace

Uncaught TypeError: Cannot read property 'theme' of undefined

```
At /Users/daviderubini/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1784

TypeError: Cannot read property 'theme' of undefined
    at Object.getFrontendUri (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1784:25)
    at HomeView.render (/packages/platformio-ide/lib/home/view.js:60:45)
    at updateSync (/packages/platformio-ide/node_modules/etch/lib/component-helpers.js:107:34)
    at /packages/platformio-ide/node_modules/etch/lib/component-helpers.js:72:7
    at ViewRegistry.performDocumentUpdate (/Applications/Atom.app/Contents/Resources/app/src/view-registry.js:238:13)
```

### Commands

```
     -0:42.3.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.17 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.7.1 
platformio-ide 2.1.0 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
tool-bar 1.1.6 
```

seksity, 2018-03-03:
also the same issue with Window x64, Atom 1.24.0 and VSCode 1.20.1, below StackTrace on Atom:-

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.24.0 x64
**Electron**: 1.6.16
**OS**: Microsoft Windows 10 Home Single Language
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.1.0

### Stack Trace

Uncaught TypeError: Cannot read property 'theme' of undefined

```
At C:\Users\Seksit\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1784

TypeError: Cannot read property 'theme' of undefined
    at Object.getFrontendUri (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1784:25)
    at HomeView.render (/packages/platformio-ide/lib/home/view.js:60:45)
    at updateSync (/packages/platformio-ide/node_modules/etch/lib/component-helpers.js:107:34)
    at /packages/platformio-ide/node_modules/etch/lib/component-helpers.js:72:7
    at ViewRegistry.performDocumentUpdate (~/AppData/Local/atom/app-1.24.0/resources/app/src/view-registry.js:238:13)
```

### Commands

```
     -0:05.2.0 platformio-ide:home (ul.list-inline.tab-bar.inset-panel)
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.17 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.7.1 
platformio-ide 2.1.0 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
pymakr 1.2.8 
tool-bar 1.1.6 
```

mblasee, 2018-03-03:
Same happened on linux, both visual studio code and atom versions are down.

1. Launch Atom
2. Launch PlatformIO Home

Atom: 1.24.0 x64
Electron: don't know how to check
OS: Ubuntu debian 16.04
Thrown From: platformio-ide package 2.1.0

**Stack Trace**
Uncaught TypeError: Cannot read property 'theme' of undefined
```
At home/blase/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1784

Uncaught` TypeError: Cannot read property 'theme' of undefined
    at Object.getFrontendUri (/home/blase/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1784:25)
    at HomeView.render (/home/blase/.atom/packages/platformio-ide/lib/home/view.js:60:45)
    at updateSync (/home/blase/.atom/packages/platformio-ide/node_modules/etch/lib/component-helpers.js:107:34)
    at /home/blase/.atom/packages/platformio-ide/node_modules/etch/lib/component-helpers.js:72:7
    at ViewRegistry.performDocumentUpdate (/usr/share/atom/resources/app/src/view-registry.js:238:13)
```
**Commands**
I don't know how to check this but if someone tells me I will complete the report. 

**Non-Core Packages**
```
autocomplete-clang@0.11.4
build@0.70.0
busy-signal@1.4.3
file-icons@2.1.17
intentions@1.1.5
language-ini@1.19.0
linter@2.2.0
linter-gcc@0.7.1
linter-ui-default@1.7.1
platformio-ide@2.1.0
platformio-ide-debugger@1.2.5
platformio-ide-terminal@2.8.0
tool-bar@1.1.6
```

hank3094, 2018-03-03:
I had this exact problem on Windows 10 x64 Atom 1.24.0 x64.

I opened PlatformIO terminal and typed "pio home". The home page came up in my browser.

After this the home page working in Atom.

I also changed the theme and syntax to Atom Dark but I don't think this has anything to do with anything.

After doing this the homepage works in Atom after restarting the program.

mblasee, 2018-03-03:
Well it fixed for me too, but I guess its not how its supposed to work.

Yozhig, 2018-03-03:
Workaround with "pio home" suggested by @hank3094 fixed this issue both for atom and vscode. Thanks.

trupntrup, 2018-03-03:
Same for me. @hank3094 thanks

mrgadotti, 2018-03-05:
I had this problem on Windows 10 x64 and Atom 1.24.0 x64.
I had similar problema running Linux Mint.

C:\Users\marce\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1784
Hide Stack Trace
TypeError: Cannot read property 'theme' of undefined
    at Object.getFrontendUri (C:\Users\marce\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1784:25)
    at HomeView.render (file:///C:/Users/marce/.atom/packages/platformio-ide/lib/home/view.js:60:45)
    at updateSync (C:\Users\marce\.atom\packages\platformio-ide\node_modules\etch\lib\component-helpers.js:107:34)
    at C:\Users\marce\.atom\packages\platformio-ide\node_modules\etch\lib\component-helpers.js:72:7
    at ViewRegistry.performDocumentUpdate (C:\Users\marce\AppData\Local\atom\app-1.24.0\resources\app\src\view-registry.js:238:13)

ivankravets, 2018-03-05:
Sorry for the issue, we did a big step ahead with multiple themes support (Dark & Light). Now all should work. Please update extension to 2.1.1

closed 2018-03-05 by ivankravets

trupntrup, 2018-03-06:
Thank you very much, it’s work.

commit f997ceeb (2018-03-05): Fix "Cannot read property 'theme' of undefined" when opening PIO Home // Resolve #1573