platformio/platformio-atom-ide #1576: Arduino autocomple not working

CawaKharkov, 2018-03-03:
OS: Win 10x64
Atom: 1.24.0
PlatformIo: 2.1.0
Clang: 
```
clang version 3.9.1 (branches/release_39) 
Target: x86_64-pc-windows-msvc 
Thread model: posix 
InstalledDir: C:\Program Files\LLVM\bin
```

`platformio.ini`
```
[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
```

Uploading to arduino and running - fine.
Base autocomplete like if,for,while - fine
**Arduino autocomplete - not working**

CawaKharkov, 2018-03-03:
And yes, I've already tried to do Rebuild project index.

ivankravets, 2018-06-21:
1. Autocompletion does not work for INO files
2. Please take a look at http://docs.platformio.org/en/latest/ide/vscode.html

closed 2018-06-21 by ivankravets