platformio/platformio-atom-ide #1581: InstallationManager

burtonscomputers, 2018-03-04:
Virtualenv Create: Traceback (most recent call last):
  File "/home/glenn/.platformio/.cache/tmp-2908173uSuapV2kM1/virtualenv-14.0.6/virtualenv.py", line 21, in <module>
    import logging
  File "/usr/lib/python2.7/logging/__init__.py", line 26, in <module>
    import sys, os, time, cStringIO, traceback, warnings, weakref, collections
  File "/usr/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

burtonscomputers, 2018-03-04:
Sorry, no idea what this all means. New install of Atom, linked with PlatformIO

ivankravets, 2018-06-19:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1298

closed 2018-06-19 by ivankravets