platformio/platformio-atom-ide #1592: InstallationManager

arturo393, 2018-03-05:
PIP:   Retrying (Retry(total=4, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Could not find a version that satisfies the requirement bottle<0.13 (from platformio==3.5.2rc3) (from versions: )
No matching distribution found for bottle<0.13 (from platformio==3.5.2rc3)

ivankravets, 2018-04-09:
What is your OS?

madleech, 2018-04-09:
Happening to me too.

**OS**
Mac OS 10.10.5

**$ atom --version**
Atom    : 1.25.1
Electron: 1.7.11
Chrome  : 58.0.3029.110
Node    : 7.9.0

**platformio-ide**
Version 2.1.3

**$ python --version**
Python 2.7.10

**$ pip --version**
pip 1.5.4 from /Library/Python/2.7/site-packages/pip-1.5.4-py2.7.egg (python 2.7)

**$ pip list**
```
altgraph (0.10.2)
arrow (0.10.0)
awscli (1.11.162)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
botocore (1.7.20)
bottle (0.12.13)
certifi (2018.1.18)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
docutils (0.14)
futures (3.1.1)
idna (2.6)
jmespath (0.9.3)
lockfile (0.12.2)
macholib (1.5.1)
matplotlib (1.3.1)
mercurial (2.9.2)
modulegraph (0.10.4)
numpy (1.8.0rc1)
pip (1.5.4)
protobuf (3.3.0)
py2app (0.7.3)
PyAL (0.2.0)
pyasn1 (0.3.6)
pygame (1.9.2a0)
pyglet (1.2.4)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
pyserial (3.4)
pysimulavr (1.0.0)
python-dateutil (2.6.1)
pytz (2013.7)
PyYAML (3.12)
requests (2.18.4)
rsa (3.4.2)
s3transfer (0.1.11)
scipy (0.13.0b1)
semantic-version (2.6.0)
setuptools (36.2.1)
SimpleParse (2.2.0)
six (1.10.0)
Twisted (13.2.0)
urllib3 (1.22)
vboxapi (1.0)
wsgiref (0.1.2)
wxPython (3.0.2.0)
wxPython-common (3.0.2.0)
xattr (0.6.4)
zope.interface (4.1.1)
```

madleech, 2018-04-09:
Could this be due to the removal of TLS < 1.2 support for pypi.python.org?

```
$ pip install -U --verbose platformio
Collecting platformio
  1 location(s) to search for versions of platformio:
  * https://pypi.python.org/simple/platformio/
  Getting page https://pypi.python.org/simple/platformio/
  Looking up "https://pypi.python.org/simple/platformio/" in the cache
  No cache entry available
  Starting new HTTPS connection (1): pypi.python.org
  "GET /simple/platformio/ HTTP/1.1" 403 170
  Status code 403 not in [200, 203, 300, 301]
  Could not fetch URL https://pypi.python.org/simple/platformio/: 403 Client Error: TLSv1.2+ is required for url: https://pypi.python.org/simple/platformio/ - skipping
  Could not find a version that satisfies the requirement platformio (from versions: )
Cleaning up...
No matching distribution found for platformio
Exception information:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/commands/install.py", line 324, in run
    requirement_set.prepare_files(finder)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/index.py", line 514, in find_requirement
    'No matching distribution found for %s' % req
DistributionNotFound: No matching distribution found for platformio
Looking up "https://pypi.python.org/pypi/pip/json" in the cache
No cache entry available
Starting new HTTPS connection (1): pypi.python.org
"GET /pypi/pip/json HTTP/1.1" 403 170
Status code 403 not in [200, 203, 300, 301]
There was an error checking the latest version of pip
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/utils/outdated.py", line 128, in pip_version_check
    resp.raise_for_status()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/_vendor/requests/models.py", line 862, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 403 Client Error: TLSv1.2+ is required for url: https://pypi.python.org/pypi/pip/json
```

dimaj, 2018-04-09:
if you guys are still having issues, I was able to overcome them on a Mac as described here: https://github.com/platformio/platformio-atom-ide/issues/1672#issuecomment-375474408

ivankravets, 2018-04-09:
@dmnbi4 @ro66s @iotmakassar @bevchou @rhodopsik @GeoSn0w @Jessie351 @sbrinkmann @pjockey @w35m17h 

Oh... I see. PyPi has removed support for old TLS, the same like Github did 1 month ago. You need to update openSSL library:

1. Install [brew](https://brew.sh/)
2. Run in terminal `brew install openssl`
3. Remove `~/.platformio` folder and restart OS/IDE.

Did it help?

Ruben-alen, 2018-04-09:
I have the problem on windows, after uninstall and install it again

ivankravets, 2018-04-09:
@Ruben-alen 

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

pjockey, 2018-04-09:
I'm at work but follow the instructions you have provided. I do know Brew is already installed.ThanksScott                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 Passport smartphone on the TELUS network.                                                                                                                                                                                                                From: Ivan KravetsSent: Monday, April 9, 2018 10:26To: platformio/platformio-atom-ideReply To: platformio/platformio-atom-ideCc: S Milliken; MentionSubject: Re: [platformio/platformio-atom-ide] No matching distribution found for bottle<0.13 (#1592)@dmnbi4 @ro66s @iotmakassar @bevchou @rhodopsik @GeoSn0w @Jessie351 @sbrinkmann @pjockey @w35m17h
Oh... I see. PyPi has removed support for old TLS, the same like Github did 1 month ago. You need to update openSSL library:

Install brew
Run in terminal brew install openssl
Remove ~/.platformio folder and restart OS/IDE.

Did it help?

madleech, 2018-04-09:
@ivankravets yes that is what I eventually ended up doing. Is there any way to catch this error and display a helpful message? I suspect you'll continue getting a lot of bug reports about this, since the error is quite un-obvious.

ivankravets, 2018-04-09:
@madleech did `brew install openssl` help you?

pjockey, 2018-04-09:
Ivan, this worked for my system. Brew and openssl were already installed,
so perhaps the deletion of the ~/.platformio directory did the trick.

Thanks

madleech, 2018-04-10:
@ivankravets I ended up `brew install`ing platformio, which by association installed an updated OpenSSL. After doing that the PlatformIO IDE installed without issue, besides the SCons issue mentioned in #1512.

Ruben-alen, 2018-04-10:
trying to install by command i get the following:

pip install -U platformio
Collecting platformio
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(0, 'unknown error (_ssl.c:3622)'),)': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(0, 'unknown error (_ssl.c:3622)'),)': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(0, 'unknown error (_ssl.c:3622)'),)': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(0, 'unknown error (_ssl.c:3622)'),)': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(0, 'unknown error (_ssl.c:3622)'),)': /simple/platformio/
  Could not fetch URL https://pypi.python.org/simple/platformio/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.python.org', port=443): Max retries exceeded with url: /simple/platformio/ (Caused by SSLError(SSLError(0, 'unknown error (_ssl.c:3622)'),)) - skipping
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio

RadioErewan, 2018-04-20:
https://stackoverflow.com/questions/49768770/not-able-to-install-python-packages-ssl-tlsv1-alert-protocol-version?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa

Updating PIP solves platformio manual install issues.

ivankravets, 2018-04-21:
We updated installer and it tries to upgrade PIP automatically for you. Try to UNINSTALL this extension and install again. It should pull new installer.

closed 2018-04-21 by ivankravets