platformio/platformio-atom-ide #1593: InstallationManager

arturo393, 2018-03-05:
PIP:   Retrying (Retry(total=4, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', BadStatusLine("''",))': /simple/bottle/
  Could not find a version that satisfies the requirement bottle<0.13 (from platformio==3.5.2rc3) (from versions: )
No matching distribution found for bottle<0.13 (from platformio==3.5.2rc3)

ivankravets, 2018-03-05:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1592

ivankravets, 2018-03-05:
What is your OS?

closed 2018-03-05 by ivankravets