platformio/platformio-atom-ide #1602: InstallationManager

1rabbit, 2018-03-07:
If you use Antivirus, it can block PlatformIO Installer. Try to disable it for a while.

Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\gerro\.platformio\.cache\tmp-6572cfj1SB4CCxSp\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\gerro\.platformio\.cache\tmp-6572cfj1SB4CCxSp\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\gerro\.platformio\.cache\tmp-6572cfj1SB4CCxSp\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\gerro\.platformio\.cache\tmp-6572cfj1SB4CCxSp\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\gerro\.platformio\.cache\tmp-6572cfj1SB4CCxSp\virtualenv-14.0.6\virtualenv.py", line 743, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Python27\Lib\subprocess.py", line 711, in __init__
    errread, errwrite)
  File "C:\Python27\Lib\subprocess.py", line 959, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

JoaoCampos89, 2018-03-08:
I already solved, I take several weeks to found out it  was the anti-virus. This #1609  says that I have to disable it and it Works. I am using Windows 10. You can add this problem to known issues can solve some headaches!

ivankravets, 2018-06-20:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1191

closed 2018-06-20 by ivankravets