platformio/platformio-atom-ide #1609: InstallationManager issue

JoaoCampos89, 2018-03-08:
InstallationManager

Virtualenv Create: 

I am trying to install platformio-ide with no success, a warning popup with  "InstallationManager Virtualenv Create:"  appears. I installed in the vs code and atom with the same problem.

My echo $PATH

echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\Python27\;C:\Python27\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\MiKTeX 2.9\miktex\bin\x64\;C:\Program Files\LLVM\bin;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files\nodejs\;C:\Users\Joao Campos\AppData\Local\Microsoft\WindowsApps;C:\MinGW\bin;C:\Users\Joao Campos\AppData\Local\Microsoft\WindowsApps;C:\Users\Joao Campos\AppData\Roaming\npm;C:\Users\Joao Campos\AppData\Local\atom\bin;C:\Program Files\Microsoft VS Code\bin

python --version
Python 2.7.14

pip --version
pip 9.0.1 from c:\python27\lib\site-packages (python 2.7)

pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
pip (9.0.1)
setuptools (28.8.0)
simpy (3.0.10)

I installed anaconda but I already uninstalled it.

JoaoCampos89, 2018-03-08:
I already solved, I take several weeks to found out it was the anti-virus. This #1602  says that I have to disable it and it Works. I am using Windows 10. You can add this problem to known issues can solve some headaches!

ivankravets, 2018-06-21:
Thank you for your solution!

closed 2018-06-21 by ivankravets