platformio/platformio-atom-ide #1623: Uncaught TypeError: Cannot read property 'dataset' of null

jcgrif, 2018-03-11:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.24.1 x64
**Electron**: 1.6.16
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.1.2

### Stack Trace

Uncaught TypeError: Cannot read property 'dataset' of null

```
At C:\Users\jgriffith\.atom\packages\platformio-ide\lib\maintenance.js:139

TypeError: Cannot read property 'dataset' of null
    at /packages/platformio-ide/lib/maintenance.js:139:56
    at Array.forEach (native)
    at Object.highlightActiveProject (/packages/platformio-ide/lib/maintenance.js:135:8)
    at doHighlight (/packages/platformio-ide/lib/main.js:227:43)
    at Function.module.exports.Emitter.simpleDispatch (~/AppData/Local/atom/app-1.24.1/resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (~/AppData/Local/atom/app-1.24.1/resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at ~/AppData/Local/atom/app-1.24.1/resources/app/src/workspace.js:550:30
```

### Commands

```
```

### Non-Core Packages

```
atom-beautify 0.32.0 
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.17 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.7.1 
platformio-ide 2.1.2 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
pros 0.7.4 
seti-icons 1.5.4 
todo-show 2.2.0 
tool-bar 1.1.6 
tool-bar-main 0.0.11 
```

jolic, 2018-12-16:
still exists with the latest version

Ofloo, 2019-08-19:
ubuntu 18.04 LTS same issue atom version 1.40.0

JeronimoProgramer, 2019-10-06:
> still exists with the latest version

He is right

JeronimoProgramer, 2019-10-06:
> [Enter steps to reproduce:]
> 
> Occurred after installing the latest update
> 
> **Atom**: 1.24.1 x64
> **Electron**: 1.6.16
> **OS**: Microsoft Windows 10 Pro
> **Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.1.2
> 
> ### Stack Trace
> Uncaught TypeError: Cannot read property 'dataset' of null
> 
> ```
> At C:\Users\jgriffith\.atom\packages\platformio-ide\lib\maintenance.js:139
> 
> TypeError: Cannot read property 'dataset' of null
>     at /packages/platformio-ide/lib/maintenance.js:139:56
>     at Array.forEach (native)
>     at Object.highlightActiveProject (/packages/platformio-ide/lib/maintenance.js:135:8)
>     at doHighlight (/packages/platformio-ide/lib/main.js:227:43)
>     at Function.module.exports.Emitter.simpleDispatch (~/AppData/Local/atom/app-1.24.1/resources/app/node_modules/event-kit/lib/emitter.js:25:20)
>     at Emitter.module.exports.Emitter.emit (~/AppData/Local/atom/app-1.24.1/resources/app/node_modules/event-kit/lib/emitter.js:141:34)
>     at ~/AppData/Local/atom/app-1.24.1/resources/app/src/workspace.js:550:30
> ```
> 
> ### Commands
> ### Non-Core Packages
> ```
> atom-beautify 0.32.0 
> autocomplete-clang 0.11.4 
> build 0.70.0 
> busy-signal 1.4.3 
> file-icons 2.1.17 
> intentions 1.1.5 
> language-ini 1.19.0 
> linter 2.2.0 
> linter-gcc 0.7.1 
> linter-ui-default 1.7.1 
> platformio-ide 2.1.2 
> platformio-ide-debugger 1.2.5 
> platformio-ide-terminal 2.8.0 
> pros 0.7.4 
> seti-icons 1.5.4 
> todo-show 2.2.0 
> tool-bar 1.1.6 
> tool-bar-main 0.0.11 
> ```

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.40.1 x64
**Electron**: 3.1.10
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.4.1

### Stack Trace

Uncaught TypeError: Cannot read property 'dataset' of null

```
At C:\Users\Jerónimo Torres\.atom\packages\platformio-ide\lib\maintenance.js:109

TypeError: Cannot read property 'dataset' of null
    at /packages/platformio-ide/lib/maintenance.js:109:57
    at Array.forEach (<anonymous>)
    at Object.highlightActiveProject (/packages/platformio-ide/lib/maintenance.js:105:8)
    at doHighlight (/packages/platformio-ide/lib/main.js:306:43)
    at Function.simpleDispatch (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:1209208)
    at Emitter.emit (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:1210649)
    at stoppedChangingActivePaneItemTimeout.setTimeout (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:425706)
```

### Commands

```
     -7:03.9.0 core:backspace (input.hidden-input)
     -6:42.4.0 core:move-up (input.hidden-input)
     -6:41.5.0 core:move-down (input.hidden-input)
  2x -6:40.8.0 editor:newline (input.hidden-input)
     -6:40.4.0 core:move-up (input.hidden-input)
  5x -6:29.4.0 core:backspace (input.hidden-input)
     -4:06.5.0 intentions:highlight (input.hidden-input)
     -4:06.2.0 core:save (input.hidden-input)
     -2:35.1.0 platformio-ide:target:build (input.hidden-input)
     -2:35.1.0 platformio-ide:target:build-F:\Proyecto CasaAutomata\HUB\MARK-I-HUB-\Hub-v0.1 (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui.teletype-Authenticated)
     -1:03.1.0 platformio-ide:settings:pkg-platformio-ide (div.tool-panel.tree-view)
```

### Non-Core Packages

```
autocomplete-clang 0.12.0 
build 0.70.0 
busy-signal 2.0.1 
colorpicker 0.3.0 
emmet 2.4.3 
file-icons 2.1.35 
intentions 1.1.5 
language-ini 1.22.0 
linter 2.3.1 
linter-gcc 0.7.1 
linter-htmlhint 1.6.2 
linter-ui-default 1.8.0 
platformio-ide 2.4.1 
platformio-ide-debugger 1.2.6 
platformio-ide-terminal 2.9.4 
teletype 0.13.4 
tool-bar 1.2.0 
```

wedgevic, 2019-11-08:
I am having that problem too

ivankravets, 2019-11-08:
Please switch to PlatformIO IDE for VSCode https://platformio.org/platformio-ide