platformio/platformio-atom-ide #1661: C/C++ project index rebuild failed

mr3188, 2018-03-20:
I¡m using platformio IDE on windows and I'm having problems with the local arduino installation libraries so I decided to use platformio IDE project libraries but I get an error when building after adding the next lines to platformio.ini in my project.

lib_deps =
  WifiManager@0.15

Error: Could not find a version that satisfies the requirement '0.15' for your system 'windows_amd64

This is the configuration file I used.

```
[platformio]
env_default = d1_mini

[env:d1_mini]
platform = espressif8266
#lib_extra_dirs = ~/Documents/Arduino/libraries
lib_deps =
WifiManager@0.15
board = d1_mini
framework = arduino
```

mr3188, 2018-03-20:
If I user WiFiManager@0.12 it works even if 0.13 already exists in  library manager

ivankravets, 2018-06-20:
There is no 0.13 version https://platformio.org/lib/show/567/WifiManager

closed 2018-06-20 by ivankravets