platformio/platformio-atom-ide #1672: No matching distribution found for bottle<0.13 (from platformio==3.5.3a5)

PianSom, 2018-03-22:
Following error generated on two different machines running OSX 10.13.3. Tried uninstall/re-install.

PIP:   Retrying (Retry(total=4, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x10385de50>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/bottle/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x10385df90>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/bottle/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x103889110>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/bottle/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x103889250>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/bottle/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x103889390>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/bottle/
  Could not find a version that satisfies the requirement bottle<0.13 (from platformio==3.5.3a5) (from versions: )
No matching distribution found for bottle<0.13 (from platformio==3.5.3a5)

dimaj, 2018-03-22:
I was able to get around this issue by installing `platformio` via `brew`
```
brew install platformio
```

Then, I've uninstalled PlatformIO from Atom and installed it again. everything worked!

EssendonPhoto, 2018-03-22:
Ah, that’s a pain - I don’t have brew on those machines. 

But thank you.

dimaj, 2018-03-22:
see if this will help you in any way: http://docs.platformio.org/en/latest/installation.html

PianSom, 2018-03-22:
Oops - wrong account! :)

PianSom, 2018-03-23:
Thanks - solved my problem.

Please close

tilakjain, 2018-03-31:
I did the same, used brew to install PlatformIO as suggested above and I could finally get it working.

ivankravets, 2018-04-09:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1592

closed 2018-04-09 by ivankravets

mkoch5, 2018-04-15:
yes worked with install over brew