platformio/platformio-atom-ide #1696: Uncaught Error: Could not start PIO Home server

yelvlab, 2018-03-29:
I've been using PIO before, but I suddenly found this mistake in the afternoon. I don't understand why. I tried to use the instructions below, and reinstall PIO.

```
Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 13, in init main
  File "d:\python27\lib\site-packages\platformio\maintenance.py", line 33, in <module>
    from platformio.managers.lib import LibraryManager
  File "d:\python27\lib\site-packages\platformio\platformio\managers\lib.py", line 402, in <module>
    def get_builtin_libs(storage_names=None):
  File "d:\python27\lib\site-packages\platformio\util.py", line 116, in __init__
    self.expire = expire / 1000  # milliseconds
TypeError: unsupported operand type(s) for /: 'function' and 'int'
Error: Traceback (most recent call last):
  File "d:\python27\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "d:\python27\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "d:\python27\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "d:\python27\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\python27\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "d:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "d:\python27\lib\site-packages\platformio\commands\home.py", line 32, in cli
    pioplus_call(sys.argv[1:])
  File "d:\python27\lib\site-packages\platformio\managers\core.py", line 134, in pioplus_call
    raise exception.ReturnErrorCode(1)
ReturnErrorCode: 1

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

    at C:\Users\yelv1\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1617:23
    at ChildProcess.onExit (C:\Users\yelv1\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:538:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:194:7)
    at ChildProcess.cp.emit (C:\Users\yelv1\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:899:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

I need help

glucee, 2018-06-05:
I have same question, don't know what happened

ivankravets, 2018-06-20:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1243

closed 2018-06-20 by ivankravets