platformio/platformio-atom-ide #1706: Virtualenv Create: Traceback (most recent call last):   File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>     main()   File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 705, in main     symlink=options.symlink)   File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment     download=download,   File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel     call_subprocess(cmd, show_stdout=False, extra_env=env)   File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess     % (cmd_desc, proc.returncode)) OSError: Command C:\Users\YUsuf\.plat...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

usuf1, 2018-03-31:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\YUsuf\.platformio\.cache\tmp-5944AqSd7gTvagX1\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\YUsuf\.plat...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

wormscoffer, 2018-05-01:
Whatever your issue is, I have the same as I get exactly the same error message.

Did  you find any resolution?

usuf1, 2018-05-02:
Yes I got some of the solution to the problem

wormscoffer, 2018-05-02:
Excellent - would you mind sharing as I have two machines and both do the same?

ivankravets, 2018-06-20:
Please upgrade to `platformio-ide` 2.1.5, should be fixed. 
Please reopen if you still need help.

closed 2018-06-20 by ivankravets