platformio/platformio-atom-ide #1711: Unable to retreive PlatformIO Core

clachance, 2018-04-02:
Hi, 

I juste made a fresh install of Atom on windows 10 and i tried to install PlatformIO-IDE, but im stucked in an infinit installing PlatformIO/ restarting Atom loop.  The thing is that when i start Atom, it still trying to retrieve the PlatformIO Core, after seconds, it says "PlatfromIO IDE has been successfully installed!" and tell me to restart atom to apply change.  If i press restart, Atom is restarting and the PlatformIO still try to install and retrieve the PlatformIO Core.

I dont have any error messages, so im completely blind on what is going on.  Anybody can help me on this one?

Christian.

ivankravets, 2018-06-21:
Should be fixed in the latest release. Please => Settings > Updates.

closed 2018-06-21 by ivankravets

clachance, 2018-06-21:
Thanks for the fix!

ivankravets, 2018-06-21:
Does it work now?