platformio/platformio-atom-ide #1734: InstallationManager

pjockey, 2018-04-08:
PIP:   Could not find a version that satisfies the requirement bottle<0.13 (from platformio==3.5.3a8) (from versions: )
No matching distribution found for bottle<0.13 (from platformio==3.5.3a8)

ivankravets, 2018-04-09:
What is your OS?

ivankravets, 2018-04-09:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1592

closed 2018-04-09 by ivankravets

pjockey, 2018-04-09:
Mac OS High SierraScott                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 Passport smartphone on the TELUS network.                                                                                                                                                                                                                From: Ivan KravetsSent: Sunday, April 8, 2018 22:43To: platformio/platformio-atom-ideReply To: platformio/platformio-atom-ideCc: S Milliken; AuthorSubject: Re: [platformio/platformio-atom-ide] InstallationManager (#1734)What is your OS?