platformio/platformio-atom-ide #1754: PlatformIO IDE on Atom makes Atom crash on quit on Mac.

senceryazici, 2018-04-13:
When I try to quit via cmd+q on Mac atom quits without a problem if the platform ide package is disabled, when I enable it Atom crashes on quit and says, "Illegal instruction: 4". After I disable it and try to quit it says 

```
Attempting to call a function in a renderer window that has been closed or released.
Function provided here: worker.js:71:22
Remote event names: destroyed, crashed
```
But it quits without a problem ? At least the Mac OS is happy about it. Otherwise I get the following (platform enabled)
[image]

I can share my settings about the package if you like. What is the problem ?

zionlion77, 2018-06-05:
I have the same problem using Atom 1.27.2 and platformio-ide 2.1.4. With Platformio enabled Atom crashes upon quit (both via CMD+Q or menu item). Any idea how to solve this issue?

zionlion77, 2018-06-05:
I just found out that I can prevent Atom from quitting unexpectedly when I have the PlatformIO home screen open. Maybe this helps.
Still, this bug is very annoying!

ivankravets, 2018-06-20:
How to reproduce this issue?

senceryazici, 2018-06-23:
To reproduce, I just enable the platformio on the settings panel, then if I try to quit the Atom editor, it crashes, no extra effort is needed to reproduce, it simply occurs when enabled.

ivankravets, 2018-11-29:
Could you reproduce this issue with the latest version?

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets