platformio/platformio-atom-ide #1760: Uncaught Error: ENOENT: no such file or directory, lstat '/@alexandrrusskih:Curtains'

skopanev, 2018-04-14:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.25.1 x64
**Electron**: 1.7.11
**OS**: Mac OS X 10.13.4
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.1.4

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, lstat '/@alexandrrusskih:Curtains'

```
At fs.js:961

Error: ENOENT: no such file or directory, lstat '/@alexandrrusskih:Curtains'
    at fs.lstatSync (fs.js:961:11)
    at Object.fs.lstatSync (ELECTRON_ASAR.js:244:16)
    at Proxy.realpathSync (fs.js:1641:21)
    at Proxy.fs.realpathSync (ELECTRON_ASAR.js:321:29)
    at /packages/platformio-ide/lib/project/helpers.js:32:25
    at getActivePioProject (/packages/platformio-ide/lib/project/helpers.js:36:6)
    at Object.highlightActiveProject (/packages/platformio-ide/lib/maintenance.js:133:13)
    at doHighlight (/packages/platformio-ide/lib/main.js:247:43)
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at stoppedChangingActivePaneItemTimeout.setTimeout (/Applications/Atom.app/Contents/Resources/app/src/workspace.js:491:26)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
console-log 1.10.0 
file-icons 2.1.18 
ide-cpp 0.0.3 
ide-json 0.2.1 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.7.1 
platformio-ide 2.1.4 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.1 
teletype 0.12.2 
tool-bar 1.1.7 
```

ivankravets, 2018-04-15:
Please uninstall and install extension again

closed 2018-04-15 by ivankravets

ivankravets, 2018-05-17:
Could you try PlatformIO IDE for VSCode? https://platformio.org/get-started/ide?install=vscode