platformio/platformio-atom-ide #1802: C/C++ project index rebuild failed

TGMods, 2018-04-27:
Hi guys,
I'm trying to build Marlin_bugfix_2.0.x for an ATMEGA2560 in PlatformIO running on Atom. My computer is running 64bit Windows 10.

I have attached the platformio.ini and an old version of platformio,ini that works as a zip file. The main difference between them is in the `lib_deps` section as substituting the library .git calls from the .zip calls enables the build to run to completion.

Is there someting I need to enable in PlatformIO to allow the use .zip files in `lib_deps` ?

```
Error: Traceback (most recent call last):
  File "c:\users\sally\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\sally\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\sally\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\sally\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\sally\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\sally\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\sally\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\sally\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "c:\users\sally\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 35, in __init__
    self._gather_tplvars()
  File "c:\users\sally\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 151, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "c:\users\sally\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 75, in get_project_build_data
    raise result.exception
IOError: [Errno 2] No such file or directory: 'C:\\3D_Printer_Firmware\\Marlin Firmware Platformio\\Marlin-bugfix-2.0.x_Azteeg_X3_Pro_D-Bot\\.piolibdeps\\_tmp_installing-_tzn6q-package\\TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57\\examples\\TMC26XMotorTester\\processing\\TMC26XMotorTest\\ChopperConfiguration.pde'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
[platformio.ini.zip](https://github.com/platformio/platformio-atom-ide/files/1954698/platformio.ini.zip)

TGMods, 2018-04-27:
UPDATE:
I have just tried the same tests on a second PC running 64-bit Windows 10 running PlatformIO on Visual Studio Code with the same results.

ivankravets, 2018-04-27:
Could you share your `platformio.ini`? Libs form archive should work, see https://github.com/platformio/platformio-core/blob/develop/tests/commands/test_lib.py#L57

TGMods, 2018-04-27:
Please have a look at the platformio.ini.zip attached to the bottom of the original post but I have attached it again. The zip file has the failing platformio.ini plus a .old version that works.
I'm using Platformio Home 0.9.5 Core 3.5.3b1. I even re-installed Atom and PlatformIO to make sure I have the latest version.

[platformio.ini.zip](https://github.com/platformio/platformio-atom-ide/files/1956131/platformio.ini.zip)

ivankravets, 2018-04-27:
Tried on macOS and Windows (VM) and could not reproduce this issue. Please remove `C:\\3D_Printer_Firmware\\Marlin Firmware Platformio\\Marlin-bugfix-2.0.x_Azteeg_X3_Pro_D-Bot\\.piolibdeps` folder and start build again

closed 2018-04-27 by ivankravets