platformio/platformio-atom-ide #1812: InstallationManager

qwert1213131, 2018-05-03:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\qwert\.platformio\.cache\tmp-13088IdvH3NdjgGjf\virtualenv-15.2.0\virtualenv.py", line 2349, in <module>
    main()
  File "C:\Users\qwert\.platformio\.cache\tmp-13088IdvH3NdjgGjf\virtualenv-15.2.0\virtualenv.py", line 712, in main
    symlink=options.symlink)
  File "C:\Users\qwert\.platformio\.cache\tmp-13088IdvH3NdjgGjf\virtualenv-15.2.0\virtualenv.py", line 927, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "C:\Users\qwert\.platformio\.cache\tmp-13088IdvH3NdjgGjf\virtualenv-15.2.0\virtualenv.py", line 1239, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\qwert\.platformio\python27\lib\shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\qwert\\.platformio\\penv\\Scripts\\python.exe'

ivankravets, 2018-06-20:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1374

ivankravets, 2018-06-20:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1900

closed 2018-06-20 by ivankravets