platformio/platformio-atom-ide #1851: C/C++ project index rebuild failed

elfringham, 2018-05-22:
Error: Traceback (most recent call last):
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/platformio/commands/init.py", line 103, in cli
    pg.generate()
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 120, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "/home/ag/.platformio/penv/local/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 126, in _render_tpl
    return bottle.template(content, **self._tplvars)
  File "/home/ag/.platformio/penv/bin/bottle.py", line 3619, in template
    return TEMPLATES[tplid].render(kwargs)
  File "/home/ag/.platformio/penv/bin/bottle.py", line 3409, in render
    self.execute(stdout, env)
  File "/home/ag/.platformio/penv/bin/bottle.py", line 3396, in execute
    eval(self.co, env)
  File "<string>", line 2, in <module>
AttributeError: 'NoneType' object has no attribute 'replace'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

elfringham, 2018-05-22:
This looks to be the same problem as reported in https://github.com/platformio/platformio-atom-ide/issues/1401 however I still get the problem despite having the suggested fix already present.

ivankravets, 2018-06-20:
Please provide an output of `platformio.ini` file.

ivankravets, 2018-07-19:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1983

closed 2018-07-19 by ivankravets