platformio/platformio-atom-ide #1863: InstallationManager

davemclaughlin, 2018-05-25:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Dave McLaughlin\.platformio\.cache\tmp-17568yTyDg9jBXO4x\virtualenv-15.2.0\virtualenv.py", line 2349, in <module>
    main()
  File "C:\Users\Dave McLaughlin\.platformio\.cache\tmp-17568yTyDg9jBXO4x\virtualenv-15.2.0\virtualenv.py", line 712, in main
    symlink=options.symlink)
  File "C:\Users\Dave McLaughlin\.platformio\.cache\tmp-17568yTyDg9jBXO4x\virtualenv-15.2.0\virtualenv.py", line 927, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "C:\Users\Dave McLaughlin\.platformio\.cache\tmp-17568yTyDg9jBXO4x\virtualenv-15.2.0\virtualenv.py", line 1239, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\Dave McLaughlin\.platformio\python27\Lib\shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\DAVEMC~1\\PLATFO~1\\penv\\Scripts\\python.exe'

ivankravets, 2018-06-20:
Please remove `C:\Users\DAVEMC1\PLATFO1\penv` folder and restart IDE

ivankravets, 2018-06-20:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1374

closed 2018-06-20 by ivankravets