platformio/platformio-atom-ide #1909: C/C++ project index rebuild failed

ilovetogetspamed, 2018-06-13:
Error: [Wed Jun 13 08:49:34 2018] Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------
Exception: [Errno 13] Permission denied: '/Users/Christos/Documents/PlatformIO/Projects/mh_simulator/.pioenvs/do-not-modify-files-here.url':
File "/Users/Christos/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 100:
PROJECTBUILD_DIR=util.get_projectbuild_dir(),
File "/Users/Christos/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 314:
raise Exception(e)
========================== [ERROR] Took 0.97 seconds ==========================

tired of having my time wasted.  I should be able to change target devices in the ini file

ilovetogetspamed, 2018-06-13:
sudo rm -rf .pioenvs/ fixed the problem.  My question is why did the permissions get wacked in the first place?  All I did was change my platform.ini from "board = uno" to "board = megaatmega2560"

ivankravets, 2018-06-20:
It's very strange. Please reopen this issue when you will have this problem again.

closed 2018-06-20 by ivankravets

ilovetogetspamed, 2018-06-20:
I had it happen again, after I opened the ticket.  Makes no sense why my permissions would get overwritten.