platformio/platformio-atom-ide #1913: mbed_config.h: no such file or directory

ThomasvdPas, 2018-06-15:
All of a sudden cc1plus.exe gives me the error that `mbed_config.h` doesn't exist.
Haven't changed anything in the code (yesterday it did compile and build), but today got the error.
This is what my `platformio.ini` looks like:
```
[env:nucleo_f446re]
platform = ststm32
board = nucleo_f446re
framework = mbed
build_flags =
  -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
  -std=c++11
targets = upload
```
The problem only occurs when running from the Platformio Ide Terminal or the standard build function (`ctrl-alt-b`) from Platformio. When I run `pio run` from the Windows cmd line on the other hand, the error doesn't occur. 

Is there somebody else with a similar problem?

Looking forward to hearing from you.

ivankravets, 2018-06-20:
Please reopen if you still need help

closed 2018-06-20 by ivankravets

ThomasvdPas, 2018-06-21:
Thanks! `pio update` fixed it.