platformio/platformio-atom-ide #1915: "Libraries" tab doesn't manage `lib_deps` libraries as expected

cinderblock, 2018-06-16:
Forgive me if this has been discussed but I did a quick search and did not find anything.

One of the reasons I like PIO over the Arduino editor is the sane library management that happens when one uses the `lib_deps` option in `platformio.ini`. The biggest headache when sharing code between machines or people is which set of libraries was used and the Arduino editor does a poor job of addressing this issue. Fortunately, the world generally has come around and the new standard of having a tool manage all the library dependencies per project is now relatively commonplace. `lib_deps` does this. `npm` does this for JavaScript. `composer` does this for PHP. I could go on, but I digress.

Now, enter the "Libraries" tab of the PlatformIO Atom extension. I was expecting this to be a gui that helped manage `lib_deps` as is what I assumed #8 was doing. Alas, it seems to be much more like the Arduino library manager that just knows how to install manually specified libraries in either global storage or a hidden directory in a selected project and it does not maintain `lib_deps`. This seems quite counter to what PIO is trying to do, imho.

To top it off, when trying to "install" a library via the "Libraries" tab, it has warnings all over the place that basically say "Use the cool `lib_deps` feature of `platformio.ini` configuration file to make managing libraries easier." without any indication that the gui tool in use isn't actually doing this for us.

Could the "Libraries" tab be brought more inline with how the PIO "Library Manager" works? I think this would help a lot of people. #369 for one.

ivankravets, 2018-06-18:
Do you mean this https://github.com/platformio/platformio-core/issues/1028?

cinderblock, 2018-06-18:
Yes, that would likely be a part of the solution to this.

One thing that might help is to remove/hide the big blue "Install" button that just does the install without actually marking the dependency. As it stands, it's basically encouraging users to go to the Arduino way of doing things.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets

cinderblock, 2020-06-14:
Yes. I moved to VSCode a while ago ;)

Still feels like PlatformIO "takes over" the editor a little too much. Often I work on non PlatformIO stuff and the build scripts can get in the way.