platformio/platformio-atom-ide #1916: Autocompletion does not work

sdouze, 2018-06-16:
I installed today atom and PlatformIO and I cannot get autocopletion for arduino to work.Any idea to solve this?
I have clang installed.
I am using atom.

ivankravets, 2018-06-20:
It does not work for INO files. http://docs.platformio.org/en/latest/ide/atom.html#intelligent-code-completion

Also, try PIO IDE for VSCode.

sdouze, 2018-06-23:
Hello,
I am using .cpp files not ino.
To use VSCode it's not a solution to the problem ,cause I don't like VSCode.

ivankravets, 2018-06-23:
What is your OS?

sdouze, 2018-06-23:
Windows 10

sdouze, 2018-06-23:
Windows 10

PNixx, 2019-03-13:
On Mac Os not working autocompletion.

ivankravets, 2019-03-13:
Please switch to PlatformIO IDE for VSCode => https://platformio.org/install/ide?install=vscode

PNixx, 2019-03-13:
I like Atom, not VSCode

ivankravets, 2019-03-13:
See http://docs.platformio.org/en/latest/ide/atom.html#intelligent-code-completion

Atom has limited support for C/C++ programming and is not recommended.

closed 2019-10-11 by sdouze