platformio/platformio-atom-ide #1924: C/C++ project index rebuild failed

mmphego, 2018-06-20:
Error: Traceback (most recent call last):
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/platformio/commands/init.py", line 103, in cli
    pg.generate()
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 120, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "/home/mmphego/.platformio/penv/local/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 126, in _render_tpl
    return bottle.template(content, **self._tplvars)
  File "/home/mmphego/.platformio/penv/bin/bottle.py", line 3619, in template
    return TEMPLATES[tplid].render(kwargs)
  File "/home/mmphego/.platformio/penv/bin/bottle.py", line 3409, in render
    self.execute(stdout, env)
  File "/home/mmphego/.platformio/penv/bin/bottle.py", line 3396, in execute
    eval(self.co, env)
  File "<string>", line 2, in <module>
AttributeError: 'NoneType' object has no attribute 'replace'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2018-06-20:
Please provide an output of `platformio.ini` file.

ivankravets, 2018-06-20:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1851

closed 2018-06-20 by ivankravets