platformio/platformio-atom-ide #1930: C/C++ project index rebuild failed

Dak-Bri, 2018-06-21:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.27.2
PIO IDE: v2.1.6
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Traceback (most recent call last):
  File "c:\users\adria\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\adria\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\adria\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\adria\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\adria\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\adria\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\adria\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\adria\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 103, in cli
    pg.generate()
  File "c:\users\adria\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 120, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "c:\users\adria\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 146, in _merge_contents
    with open(dst_path, "w") as f:
IOError: [Errno 13] Permission denied: 'C:\\Users\\adria\\Proyectos\\intelliheating_local\\SensorPresencia\\v2\\.clang_complete'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

ivankravets, 2018-06-21:
Could you remove this file `C:\\Users\\adria\\Proyectos\\intelliheating_local\\SensorPresencia\\v2\\.clang_complete` manually and restart IDE? Do you see this errors again?

Dak-Bri, 2018-06-22:
Done and works perfectly. 

Thank you!!!!

Why this error happens?

ivankravets, 2018-06-25:
It seems that Atom's clang process hung on...

closed 2018-06-25 by ivankravets