platformio/platformio-atom-ide #1931: InstallationManager

alexgithub56, 2018-06-21:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.27.2
PIO IDE: v2.1.6
System: Windows_NT, 10.0.17134, x64

# Exception
```
Virtualenv Create: ImportError: No module named site

```

ivankravets, 2018-06-22:
How do you resolve it?

alexgithub56, 2018-06-23:
I can not fix it and not understand document so I think I don't have experience for this software. I just try to use it. So, in this case, I decide to change to another software.

ivankravets, 2018-06-23:
Could you provide an output for commands in https://github.com/platformio/platformio-atom-ide/issues/1931#issuecomment-399036903 ?

alexgithub56, 2018-06-26:
> Could you provide an output for commands in #1931 (comment)
> ?

Hi Ivan Kravets,

1. I used Windows 10 Home 64 bits.

2. I download file "AtomSetup-x64.exe" and install as administrator.

3 I download file "python-2.7.15.amd64.msi" and install for all users to
"c:\python27\".

4. I add "C:\Python27\" and "C:\Python27\Scripts" path to system variables.

5. I start Atom as administrator and install package name "platform-ide
2.1.7".

6. PlatformIO IDE installing and got error:
User's Virtualenv: Traceback (most recent call last): File
"C:\Python27\Lib\site-packages\virtualenv.py", line 2343, in main() File
"C:\Python27\Lib\site-packages\virtualenv.py", line 712, in main
symlink=options.symlink) File
"C:\Python27\Lib\site-packages\virtualenv.py", line 927, in
create_environment site_packages=site_packages, clear=clear,
symlink=symlink)) File "C:\Python27\Lib\site-packages\virtualenv.py", line
1134, in install_python copy_required_modules(home_dir, symlink) File
"C:\Python27\Lib\site-packages\virtualenv.py", line 1064, in
copy_required_modules dst_filename = change_prefix(filename, dst_prefix)
File "C:\Python27\Lib\site-packages\virtualenv.py", line 1038, in
change_prefix (filename, prefixes) AssertionError: Filename
C:\Python27\Lib\os.py does not start with any of these prefixes:
['C:\python27']

7. For your advice, this is my informations:
C:\WINDOWS\system32>echo %PATH%
C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\Program
Files (x86)\iis
express\PHP\v5.3;C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\Program
Files (x86)\iis
express\PHP\v5.3;%PATH%;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program
Files (x86)\Nmap;C:\Program Files\Docker Toolbox;C:\Users\Anurath
Sattabongkot\AppData\Local\Microsoft\WindowsApps;C:\PHP;C:\Users\Anurath
Sattabongkot\AppData\Local\Programs;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program
Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI
Foundation\VISA\Win64\Bin\;C:\Program
Files\Git\cmd;;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program
Files (x86)\Nmap;C:\Program Files\Docker Toolbox;C:\Users\Anurath
Sattabongkot\AppData\Local\Microsoft\WindowsApps;C:\PHP;C:\Users\Anurath
Sattabongkot\AppData\Local\Programs;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program
Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI
Foundation\VISA\Win64\Bin\;C:\Program Files\Git\cmd;

C:\WINDOWS\system32>echo %PYTHONPATH%
C:\Python27\DLLs;C:\Python27\Lib;C:\Python27\Lib\site-packages

C:\WINDOWS\system32>python --version
Python 2.7.15

C:\WINDOWS\system32>pip --version
pip 10.0.1 from C:\Python27\Lib\site-packages\pip (python 2.7)

C:\WINDOWS\system32>pip list
Package    Version
---------- -------
pip        10.0.1
setuptools 39.0.1
virtualenv 16.0.0

Regards,
Anurath