platformio/platformio-atom-ide #1935: Add custom python path

masoudr, 2018-06-22:
Hi,
I'm using Atom with python tools which all of them are built based on `python3` on Windows.
`python` in my system variable refers to python3. After I installed `pio`. All my Atom packages became broken and the python path forced to `python2.7`. I'm not sure if this problem is related or not, but is there any option to force `pio` to use a custom path for python2.7?
Thanks.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets

masoudr, 2020-06-14:
Thanks for the update.
I'm using VScode but the problem still exists. If I don't add the PIO's script directory to my PATH, I can compile my code, but other features like Debugging need the PIO's script directory to be added and if I do it will mess up my python environment.