platformio/platformio-atom-ide #1940: Could not start PIO Home server

TheStonePlayer, 2018-06-24:
# Description of problem

Uncaught Error: Could not start PIO Home server: Traceback (most recent call last): File "main.py", line 16, in init main File "cmd_debug.py", line 24, in init cmd_debug File "C:\Users\26492.platformio\packages\contrib-pysite\twisted\internet\reactor.py", line 39, in default.install() File "C:\Users\26492.platformio\packages\contrib-pysite\twisted\internet\selectreactor.py", line 196, in install reactor = SelectReactor() File "C:\Users\26492.platformio\packages\contrib-pysite\twisted\internet\selectreactor.py", line 72, in init posixbase.PosixReactorBase.init(self) File "C:\Users\26492.platformio\packages\contrib-pysite\twisted\internet\base.py", line 515, in init self.installWaker() File "C:\Users\26492.platformio\packages\contrib-pysite\twisted\internet\posixbase.py", line 284, in installWaker self.waker = self._wakerFactory(self) File "C:\Users\26492.platformio\packages\contrib-pysite\twisted\internet\posixbase.py", line 79, in init client.connect(server.getsockname()) File "K:\Python27\Lib\socket.py", line 228, in meth return getattr(self._sock,name)(*args) socket.error: [Errno 10060]

Stack trace

Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\26492\.platformio\packages\contrib-pysite\twisted\internet\reactor.py", line 39, in <module>
    default.install()
  File "C:\Users\26492\.platformio\packages\contrib-pysite\twisted\internet\selectreactor.py", line 196, in install
    reactor = SelectReactor()
  File "C:\Users\26492\.platformio\packages\contrib-pysite\twisted\internet\selectreactor.py", line 72, in __init__
    posixbase.PosixReactorBase.__init__(self)
  File "C:\Users\26492\.platformio\packages\contrib-pysite\twisted\internet\base.py", line 515, in __init__
    self.installWaker()
  File "C:\Users\26492\.platformio\packages\contrib-pysite\twisted\internet\posixbase.py", line 284, in installWaker
    self.waker = self._wakerFactory(self)
  File "C:\Users\26492\.platformio\packages\contrib-pysite\twisted\internet\posixbase.py", line 79, in __init__
    client.connect(server.getsockname())
  File "K:\Python27\Lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 10060] 
    at C:\Users\26492\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:25739
    at ChildProcess.m (C:\Users\26492\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:6548)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at ChildProcess.cp.emit (C:\Users\26492\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:925:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)

# Configuration
Atom: 1.28.0
PIO IDE: v2.1.6
System: Windows_NT, 10.0.15063, x64

# Exception
```
Could not start PIO Home server: Error: timeout
```

ivankravets, 2018-06-25:
Do you have a network firewall in a system?

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets