platformio/platformio-atom-ide #1946: InstallationManager

jcgrif, 2018-06-25:
# Description of problem
Been having issues with PlatformIO, uninstalled PlatformIO then reinstalled.  Installed Python 2.7 for my user as an attempt to resolve.  Now getting this error.  Tried running Atom as local administrator in attempt to resolve.  Still no success.

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.28.0
PIO IDE: v2.1.6
System: Windows_NT, 10.0.17134, x64

# Exception
```
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\jgriffith\.platformio\.cache\tmp-1812880ak9ydp4y4x\virtualenv-16.0.0\virtualenv.py", line 2343, in <module>
    main()
  File "C:\Users\jgriffith\.platformio\.cache\tmp-1812880ak9ydp4y4x\virtualenv-16.0.0\virtualenv.py", line 712, in main
    symlink=options.symlink)
  File "C:\Users\jgriffith\.platformio\.cache\tmp-1812880ak9ydp4y4x\virtualenv-16.0.0\virtualenv.py", line 927, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "C:\Users\jgriffith\.platformio\.cache\tmp-1812880ak9ydp4y4x\virtualenv-16.0.0\virtualenv.py", line 1276, in install_python
    shutil.copyfile(pythondll, os.path.join(os.path.dirname(py_executable), py_executable_dll))
  File "C:\Python27\lib\shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\jgriffith\\.platformio\\penv\\Scripts\\python27.dll'

```

jcgrif, 2018-06-25:
Tried deleting C:\users\jgriffith\.platormio\penv as refered in other tickets, and now when I try to create a new project the Board: field in the dialog just spins and the Framework dropdown never populates.  The spinner on the Home page under recent news, never stops either.

ivankravets, 2018-06-25:
1. Please upgrade this package to v2.1.8 via Atom > Settings > Updates 
2. Please navigate to Windows > Control Panel > Installed Programs and remove all Pythons.
3. Please remove `C:\\Users\\jgriffith\\.platformio\\penv` and `C:\\Users\\jgriffith\\.platformio\\python27` folders.

Restart Atom. Should work.

closed 2018-06-25 by ivankravets

ivankravets, 2018-06-26:
I've just installed Python3 and PIO. Everything works well.

1. Please uninstall all Pythons from a system (Control Panel > Apps > uninstall)
2. Remove `C:\\Users\\jgriffith\\.platformio\\penv` and `C:\\Users\\jgriffith\\.platformio\\python27` folder
3. Restart Atom.

Please navigate to Atom > Settings > Packages > `platformio-ide` and select using built-in PIO Core.

jcgrif, 2018-06-27:
Still the same thing.    Atom > Settings > Packages > platformio-ide and select using built-in PIO Core was already set to use the built-in PIO Core.

I tried running the Upgrade PIO Core and got the following
You're up-to-date!
PlatformIO 3.6.0a3 is currently the newest version available.

Then tried upgrading packages, platforms, and libraries and got the following.
Uninstalling tool-pioplus @ 1.3.1:      Please manually remove file `C:\Users\jgriffith\.platformio\packages\tool-pioplus\pioplus.exe`
Error: Traceback (most recent call last):
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)

  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\platformio\commands\update.py", line 35, in cli
    update_core_packages(only_check)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 95, in update_core_packages
    pm.update(name, requirements, only_check=only_check)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 56, in update
    result = PackageManager.update(self, *args, **kwargs)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 802, in update
    self.uninstall(pkg_dir, after_update=True)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 743, in uninstall
    util.rmtree_(pkg_dir)
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\platformio\util.py", line 787, in rmtree_
    return rmtree(path, onerror=_onerror)
  File "C:\Users\jgriffith\.platformio\python27\Lib\shutil.py", line 266, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "c:\users\jgriffith\.platformio\penv\lib\site-packages\platformio\util.py", line 785, in _onerror
    raise e
WindowsError: [Error 5] Access is denied: 'C:\\Users\\jgriffith\\.platformio\\packages\\tool-pioplus\\pioplus.exe'

============================================================

I exited Atom, had to kill the hung pioplus process and deleted the pioplus.exe file, the penv and python27 directories from under the .platformio directory.  The restarted Atom.  It went through the PlatformIDE: Installing dialog yet again.

This spinning Icon on the board selection still occurs.  But when I try to do an Update Packages, Platforms, and Updates, I get the following:

PS C:\Users\jgriffith\.atom> platformio update
Updating contrib-pysite                  @ 0.3.0          [Up-to-date]
Updating contrib-piohome                 @ 0.9.6          [Up-to-date]
Updating tool-pioplus                    @ 1.3.3          [Up-to-date]

Platform Manager
================

Library Manager
===============
PS C:\Users\jgriffith\.atom>

If I do an update Atom Packages I get the message that "All of your packages are up to date!"

Platform-ide in the package list still reports version 2.1.7.  Is there a newer version.

ivankravets, 2018-06-27:
Please open PIO IDE Terminal and type `pio boards`. Do you see any output?

jcgrif, 2018-06-27:
So, I found that there was still a C:\Python27 on my path.  Even after Uninstalling Python27 and Python3.6.4.  So when I was letting platformio go through its re-initialization process after doing your instructions,  it was recreating the C:\Python even though I had Uninstalled it and deleted it.  Killed the hung pioplus.exe process (It wouldn't let me do the deletes because of inuse files) So, I removed the C:\Python27 from the path.  Deleted the C:\Python27 directory again, and deleted the entire C:\Users\jgriffith\.platformio directory.  Rebooted the system.  Then started Atom.  It went through its install process again, and **_now it works._**  It seems that because C:\Python27 was on the path, that each time it did the install that the directory was getting recreated and at least part of Python27 was getting installed there.

Thanks for your help.

ivankravets, 2018-06-27:
Hm.. We don't create `C:\Python27` folder. All things which we do they are among `.platformio` folder which is located in your home dir.